GET-RAHS/2022-2024_Sockeye #18: Canopy stiffening

Hampsterkeed, 2023-10-25:
canopy flops around when lifted, fix.

MatthewF0ng, 2024-01-12:
One would need to add metal bars inside the canopy. Also, the locking mechanism is a bit difficult to push down on and doesn't make the canopy flush with the body. The hinges can be tightened for better alignment.

MatthewF0ng, 2024-01-22:
[image]
Small gap not optimal
[image]
Both sections of the lock are not aligned properly
https://mail.google.com/mail/u/1?ui=2&ik=701bd3c312&attid=0.1&permmsgid=msg-a:r-6039258511477720978&th=18d31ffb6e3f205a&view=att&disp=safe&realattid=18d31fda1bf4b6178373
Alignment pieces stopping the canopy from locking

MatthewF0ng, 2024-01-22:
It would be nice if someone else could help.